openhwfoundation/cva6-sdk #18: v0.3.0 riscv-tests/benchmarks compilation failures

Steinegger, 2019-06-17:
Hello,

When running `make all` using the v0.3.0 (366b617) release, some benchmarks and/or tests are not compiled. It appears as though some make-targets were removed compared to the v0.2.0 release and `riscv64-unknown-elf-gcc` is now missing. The logfile shows the following:

```
make[1]: Entering directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-isa-sim/build'
../scripts/mk-install-dirs.sh /workstore/user/ariane-4.2/toolchain/include/spike
mkdir /workstore/user/ariane-4.2/toolchain/include/spike
for file in ../riscv/common.h ../riscv/decode.h ../riscv/devices.h ../riscv/disasm.h ../riscv/dts.h ../riscv/mmu.h ../riscv/processor.h ../riscv/sim.h ../riscv/simif.h ../riscv/trap.h ../riscv/encoding.h ../riscv/cachesim.h ../riscv/memtracer.h ../riscv/tracer.h ../riscv/extension.h ../riscv/rocc.h ../riscv/insn_template.h ../riscv/mulhi.h ../riscv/debug_module.h ../riscv/debug_rom_defines.h ../riscv/remote_bitbang.h ../riscv/jtag_dtm.h icache.h insn_list.h ../softfloat/internals.h ../softfloat/platform.h ../softfloat/primitives.h ../softfloat/primitiveTypes.h ../softfloat/softfloat.h ../softfloat/softfloat_types.h ../softfloat/specialize.h config.h; \
  do \
    /usr/bin/install -c -m 444 $file /workstore/user/ariane-4.2/toolchain/include/spike; \
  done
../scripts/mk-install-dirs.sh /workstore/user/ariane-4.2/toolchain/lib
for file in libriscv.so libdummy_rocc.so libsoftfloat.so libspike_main.so; \
  do \
    /usr/bin/install -c -m 644 $file /workstore/user/ariane-4.2/toolchain/lib; \
  done
../scripts/mk-install-dirs.sh /workstore/user/ariane-4.2/toolchain/bin
for file in spike spike-dasm xspike termios-xspike; \
  do \
    /usr/bin/install -c -m 555 $file /workstore/user/ariane-4.2/toolchain/bin; \
  done
../scripts/mk-install-dirs.sh /workstore/user/ariane-4.2/toolchain/lib/pkgconfig/
for file in riscv-riscv.pc riscv-dummy_rocc.pc riscv-softfloat.pc riscv-spike_main.pc; \
  do \
    /usr/bin/install -c -m 444 $file /workstore/user/ariane-4.2/toolchain/lib/pkgconfig/; \
  done
make[1]: Leaving directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-isa-sim/build'
mkdir -p riscv-tests/build
cd riscv-tests/build;\
autoconf;\
../configure --prefix=/workstore/user/ariane-4.2/toolchain/target;\
make -j20;\
make install;\
cd /workstore/user/ariane-4.2/ariane-sdk
autoconf: error: no input file
checking for gcc... gcc
configure: WARNING: using cross tools not prefixed with host triplet
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
configure: creating ./config.status
config.status: creating Makefile
make[1]: Entering directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build'
mkdir -p benchmarks
mkdir -p isa
make -C benchmarks -f /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/Makefile src_dir=/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks XLEN=64
make -C isa -f /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/Makefile src_dir=/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa XLEN=64
make[2]: Entering directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/benchmarks'
riscv64-unknown-elf-gcc -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/../env -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/common -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/median -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/qsort -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/rsort -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/towers -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/vvadd -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/multiply -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/mm -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/dhrystone -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/spmv -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/mt-vvadd -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/mt-matmul -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o median.riscv /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/median/median_main.c /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/median/median.c /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/common/syscalls.c /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/common/test.ld
make[2]: riscv64-unknown-elf-gcc: Command not found
/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../benchmarks/Makefile:54: recipe for target 'median.riscv' failed
make[2]: *** [median.riscv] Error 127
make[2]: Leaving directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/benchmarks'
Makefile:24: recipe for target 'benchmarks' failed
make[1]: *** [benchmarks] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Entering directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/isa'
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/../env/p -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/macros/scalar -T/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/../env/p/link.ld /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/rv32ui/simple.S -o rv32ui-p-simple
make[2]: riscv64-unknown-elf-gcc: Command not found
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/../env/p -I/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/macros/scalar -T/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/../env/p/link.ld /workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/rv32ui/add.S -o rv32ui-p-add
make[2]: riscv64-unknown-elf-gcc: Command not found
/workstore/user/ariane-4.2/ariane-sdk/riscv-tests/build/../isa/Makefile:74: recipe for target 'rv32ui-p-simple' failed
make[2]: *** [rv32ui-p-simple] Error 127
make[2]: *** Waiting for unfinished jobs....
[... 19 lines cut ...]
cd riscv-pk/build;\
../configure --prefix=/workstore/user/ariane-4.2/toolchain --host=riscv64-unknown-linux-gnu-elf CC=riscv64-unknown-linux-gnu-gcc OBJDUMP=riscv64-unknown-linux-gnu-objdump;\
make -j20;\
make install;\
cd /workstore/user/ariane-4.2/ariane-sdk
checking build system type... x86_64-pc-linux-gnu
checking host system type... Invalid configuration `riscv64-unknown-linux-gnu-elf': machine `riscv64-unknown-linux-gnu' not recognized
configure: error: /bin/sh ../scripts/config.sub riscv64-unknown-linux-gnu-elf failed
make[1]: Entering directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-pk/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-pk/build'
make[1]: Entering directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-pk/build'
make[1]: *** No rule to make target 'install'.  Stop.
make[1]: Leaving directory '/workstore/user/ariane-4.2/ariane-sdk/riscv-pk/build'
```

Steps to reproduce:
```
export RISCV=$(pwd)/toolchain
export PATH=$PATH:$RISCV/bin
git clone --branch v0.3.0 https://github.com/pulp-platform/ariane-sdk.git 
cd ariane-sdk
git submodule update --init --recursive
make all
```

Steinegger, 2019-07-15:
I still get the same error with commit ff9d898 when running `make all`, do I have to run something else manually?
I did fresh checkout and tried it on two different machines.

<details>
 <summary>Logfile</summary>

```
make -C benchmarks -f /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/Makefile src_dir=/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks XLEN=64 
make -C isa -f /home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa/Makefile src_dir=/home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa XLEN=64 
make[2]: Entering directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build/benchmarks'
riscv64-unknown-elf-gcc -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/../env -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/median -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/qsort -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/rsort -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/towers -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/vvadd -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/multiply -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/mm -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/dhrystone -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/spmv -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/mt-vvadd -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/mt-matmul -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o median.riscv /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/median/median_main.c /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/median/median.c /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common/syscalls.c /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common/test.ld
make[2]: riscv64-unknown-elf-gcc: Command not found
make[2]: *** [/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/Makefile:54: median.riscv] Error 127
make[2]: Leaving directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build/benchmarks'
make[1]: *** [Makefile:25: benchmarks] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Entering directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build/isa'
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa/../env/p -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa/macros/scalar -T/home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa/../env/p/link.ld /home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa/rv32ui/simple.S -o rv32ui-p-simple
make[2]: riscv64-unknown-elf-gcc: Command not found
make[2]: *** [/home/user/tempstuff/ariane-sdk/riscv-tests/build/../isa/Makefile:74: rv32ui-p-simple] Error 127
make[2]: Leaving directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build/isa'
make[1]: *** [Makefile:29: isa] Error 2
make[1]: Leaving directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build'
make[1]: Entering directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build'
mkdir -p benchmarks
make -C benchmarks -f /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/Makefile src_dir=/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks XLEN=64 
make[2]: Entering directory '/home/user/tempstuff/ariane-sdk/riscv-tests/build/benchmarks'
riscv64-unknown-elf-gcc -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/../env -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/median -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/qsort -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/rsort -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/towers -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/vvadd -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/multiply -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/mm -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/dhrystone -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/spmv -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/mt-vvadd -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/mt-matmul -I/home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o median.riscv /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/median/median_main.c /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/median/median.c /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common/syscalls.c /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T /home/user/tempstuff/ariane-sdk/riscv-tests/build/../benchmarks/common/test.ld
make[2]: riscv64-unknown-elf-gcc: Command not found
[... cut ...]
make[1]: Entering directory '/home/user/tempstuff/ariane-sdk/riscv-pk/build'
make[1]: *** No targets specified and no makefile found. Stop.
make[1]: Leaving directory '/home/user/tempstuff/ariane-sdk/riscv-pk/build'
make[1]: Entering directory '/home/user/tempstuff/ariane-sdk/riscv-pk/build'
make[1]: *** No rule to make target 'install'. Stop.
make[1]: Leaving directory '/home/user/tempstuff/ariane-sdk/riscv-pk/build'
make all  8145.41s user 778.35s system 560% cpu 26:30.82 total
```

</details>

Iripi97, 2019-08-05:
I am also getting this error too. Is there still no fix or idea how to avoid this behavior?

Moschn, 2019-08-06:
What is the content of the environment variables `$RISCV` and `$PATH`?

Steinegger, 2019-08-06:
```
% echo $RISCV
/workstore/ssteinegger/ariane-4.2/toolchain

% echo $PATH
/workstore/ssteinegger/verilator/bin:/workstore/ssteinegger/ariane-4.2/toolchain/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/gurobi/linux64/bin:/opt/anaconda3/bin:/usr/local/cuda-10.0/bin
```

Before compiling the project I removed the checked out repository. The compilation doesn't stop at the error, so some scrolling is necessary to see the error in the output.

Moschn, 2019-08-06:
And `riscv64-unknown-elf-gcc` exists in your path?

Steinegger, 2019-08-06:
it looks like riscv64-unknown-elf-gcc isn't created, but instead riscv64-unknown-linux-gnu-*

```
% ls -1 $RISCV/bin                                         
elf2hex
riscv64-unknown-linux-gnu-addr2line
riscv64-unknown-linux-gnu-ar
riscv64-unknown-linux-gnu-as
riscv64-unknown-linux-gnu-cpp
riscv64-unknown-linux-gnu-c++
riscv64-unknown-linux-gnu-c++filt
riscv64-unknown-linux-gnu-elfedit
riscv64-unknown-linux-gnu-gcc
riscv64-unknown-linux-gnu-gcc-8.2.0
riscv64-unknown-linux-gnu-gcc-ar
riscv64-unknown-linux-gnu-gcc-nm
riscv64-unknown-linux-gnu-gcc-ranlib
riscv64-unknown-linux-gnu-gcov
riscv64-unknown-linux-gnu-gcov-dump
riscv64-unknown-linux-gnu-gcov-tool
riscv64-unknown-linux-gnu-gfortran
riscv64-unknown-linux-gnu-gprof
riscv64-unknown-linux-gnu-g++
riscv64-unknown-linux-gnu-ld
riscv64-unknown-linux-gnu-ld.bfd
riscv64-unknown-linux-gnu-nm
riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-objdump
riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-strip
spike
spike-dasm
termios-xspike
xspike
```

and 

```
% ls -1 $RISCV
bin
include
lib
libexec
riscv64-unknown-linux-gnu
share
sysroot
```

Moschn, 2019-08-06:
Thanks! I just managed to reproduce your problem...

Moschn, 2019-08-06:
Can you try the changes in the new PR and confirm that your problem is fixed?

Iripi97, 2019-08-09:
After making the suggested changes (to the Makefile) where the output fails it says:
"checking host system type... Invalid configuration 'riscv64-unknown-linux-gnu-elf': machine 'riscv64-unknown-linux-gnu' not recognized
configure: error: /bin/sh ../scripts/config.sub riscv64-unknown-linux-gnu-elf failed
make[1]: Entering directory '/home/justin/SDK/ariane_sdk/riscv-pk/build'
make[1]: *** No targets specified and no makefile found. Stop."
Essentially the last part that is shown in the initial comment on this issue. No change with PR.

Moschn, 2019-08-12:
I just tried again on my machine, and there is no error. Did you run `make clean-all` before rebuilding?

Iripi97, 2019-08-13:
@Moschn I tried a fresh download of the repository just to be sure. The error still persists. It appears to be an issue with using the riscv64-unknown-linux-gnu for the riscv-pk tools. Should I try changing the Makefile section for riscv-pk as we did for the riscv-tests section (as described above)?

Moschn, 2019-08-13:
Ah this is an error in a different target. Thanks for reporting. I pushed a fix to the same branch. Can you verify that the error is gone?

Iripi97, 2019-08-13:
@Moschn thank you very much for the help. The issue is resolved the tools are built successfully with the added changes!

commit ff9d898c (2019-07-10): Fix #18 (#19)
PR #20 (2019-08-14): Fix riscv-tests dependencies
Fixes #18
commit 9d850b9d (2019-08-14): Fix riscv-tests dependencies

Fixes #18
commit 5881a971 (2020-05-07): Fix #18 (#19)
closed 2019-08-14 by Moschn