openhwfoundation/cva6-sdk #21: make vmlinux issue

mathmax12, 2019-08-12:
Hi,
I am learning how to generate the Linux image by myself with the ariane-sdk, but got an error.

I already installed the riscv-gnu-toolchain, riscv-tools and riscv-pk.
And have successfully run the hello workd.c
https://github.com/pulp-platform/ariane#running-user-space-applications

So I think I don't need to run "make all" under the ariane-sdk folder again to reinstall the above tools.

After set up the environment with 
$ export RISCV=/opt/riscv
$ export PATH=$PATH:$RISCV/bin

When I try to run "make vmlinux", I got the following error:
`mkdir -p /opt/riscv
mkdir -p riscv-gnu-toolchain/build
cd riscv-gnu-toolchain/build;\
        ../configure --prefix=/opt/riscv;\
        cd /home/riscv_project/ariane-sdk
/bin/sh: 2: ../configure: Permission denied
cd riscv-gnu-toolchain/build;\
        make -j88;\
        cd /home/riscv_project/ariane-sdk
make[1]: Entering directory '/home/riscv_project/ariane-sdk/riscv-gnu-toolchain/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/home/riscv_project/ariane-sdk/riscv-gnu-toolchain/build'
mkdir -p riscv-gnu-toolchain/build
cd riscv-gnu-toolchain/build;\
../configure --prefix=/opt/riscv --disable-gdb;\
cd /home/riscv_project/ariane-sdk
/bin/sh: 2: ../configure: Permission denied
cd riscv-gnu-toolchain/build;\
make linux -j88;\
cd /home/riscv_project/ariane-sdk
make[1]: Entering directory '/home/riscv_project/ariane-sdk/riscv-gnu-toolchain/build'
make[1]: *** No rule to make target 'linux'.  Stop.
make[1]: Leaving directory '/home/riscv_project/ariane-sdk/riscv-gnu-toolchain/build'
cd ./cachetest/ && /opt/riscv/bin/riscv64-unknown-linux-gnu-gcc cachetest.c -o cachetest.elf
cp ./cachetest/cachetest.elf rootfs/
`
Did I miss somthing?
Thanks

Moschn, 2019-08-13:
Is your riscv toolchain installed in `$RISCV/bin/riscv-unknown-elf-gcc` and `$RISCV/bin/riscv-unknown-linux-gnu-gcc`? You need both toolchains: newlib and linux-gnu.

It seems like one of those is missing and the script tries to install it. However, the git submodule for `riscv-gnu-toolchain` seems to be missing. Did you pull all the submodules?

mathmax12, 2019-08-13:
Hi Moschn,
Thanks for the reply.
Here are the compilers under my riscv/bin
[image]

I followed the instructions to install the riscv-gnu-toolchain and think it's ok.

I tried to run the "make all **("all: gnu-toolchain-libc fesvr isa-sim tests pk".)"** under the ariane-sdk and rebuild it. 
During the make pk stage I met
"
mkdir -p riscv-pk/build
cd riscv-pk/build;\
../configure --prefix=/opt/riscv --host=riscv64-unknown-linux-gnu-elf CC=riscv64-unknown-linux-gnu-gcc OBJDUMP=riscv64-unknown-linux-gnu-objdump;\
make -j88;\
make install;\
cd /home/riscv_project/temp/ariane-sdk
checking build system type... x86_64-pc-linux-gnu
checking host system type... Invalid configuration `riscv64-unknown-linux-gnu-elf': machine `riscv64-unknown-linux-gnu' not recognized
configure: error: /bin/bash ../scripts/config.sub riscv64-unknown-linux-gnu-elf failed
make[1]: Entering directory '/home/riscv_project/temp/ariane-sdk/riscv-pk/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/home/riscv_project/temp/ariane-sdk/riscv-pk/build'
make[1]: Entering directory '/home/riscv_project/temp/ariane-sdk/riscv-pk/build'
make[1]: *** No rule to make target 'install'.  Stop.
make[1]: Leaving directory '/home/riscv_project/temp/ariane-sdk/riscv-pk/build'
root@msl-mip-dev18:/home/riscv_project/temp/ariane-sdk$make pk
"
waht may cause the error?
Thanks.

mathmax12, 2019-08-13:
I just notice that the configureation of the makefile in the Ariane-SDK for pk is different from that of the makefile in pk folder.

Makefile in Ariane-sdk: 
pk-co                 = --prefix=$(RISCV) **--host=riscv64-unknown-linux-gnu-elf CC=riscv64-unknown-**linux-gnu-gcc OBJDUMP=riscv64-unknown-linux-gnu-objdump 

Makefile in Ariane-sdk: 
../configure --prefix=$RISCV **--host=riscv64-unknown-elf**

If I change the confiure to the $RISCV **--host=riscv64-unknown-elf** then the make all in ariane-sdk can be passed.

Is that correct?

Moschn, 2019-08-14:
I think you are hitting issue #18. I already have a fix for that in #20 but it seems like your solution is also correct.

closed 2019-08-22 by Moschn