openhwfoundation/cva6-sdk #23: Linux boot get stuck

mathmax12, 2019-08-22:
I post this issue under Ariane but can't find a way to delete it there.
Can I get help with this issue?
https://github.com/pulp-platform/ariane/issues/309

Thanks.

Moschn, 2019-08-23:
Can you try to use the linux image from the [releases](https://github.com/pulp-platform/ariane-sdk/releases)?

mathmax12, 2019-08-23:
The released Linux image(4.2) works.
Is the flow used for this image is different from the one in the readme of ariane-sdk?
I'd like to make this image-building flow work in case I need to modify the kernel.

Thanks

Moschn, 2019-08-26:
No, the build flow is the same. I have no idea why it would not work. Maybe @zarubaf can help?

zarubaf, 2019-10-09:
Hm, no idea actually. Especially as it boots relatively far.

mathmax12, 2019-10-09:
Thanks anyway!
I switched to lowerisc-chip and found that the build bbl works.

closed 2019-10-09 by mathmax12

Kendidi, 2020-08-06:
@mathmax12 

Hi, I am trying to load bbl with gdb to Ariane on a FPGA and hope you or someone can give me some pointers. 
DDR on my board starts from 0x200000000 for 8GB. I have modified PAGE_OFFSET to 0xfffffffe00000000 (2's complement of 0x200000000) when I built vmlinux.  After I built bbl, I loaded bbl to address 0x280000000 with riscv64-unknown-elf-gdb. I wonder what should I do next to have it run, to copy code to 0x20000000, etc.  Thanks a lot in advance!