openhwfoundation/cva6-sdk #24: Incorrect selection of kernel headers: expected 4.13.x, got 5.0.x

mathmax12, 2019-10-10:
Hi,

It's a little wired.
The "make vmlinux" work well before.
But I got a "Incorrect selection of kernel headers: expected 4.13.x, got 5.0.x" issue when run "make vmlinux".
I  tried to change the both of "buildroot -> Toolchain -> Custom kernel headers series " and kernel version to 4.18, which is the latest version in my buidroot.

But I still got the issue  "Incorrect selection of kernel headers: expected 4.18.x, got 5.0.x".

Any help is appreciated.

Moschn, 2019-10-10:
Did you recently update your host kernel? I think the headers in toolchain->kernel headers are the ones supported by your cross compilation toolchain. Therefore they must be the same as your host kernel. 

I would assume that we need to bump the buildroot submodule every once in a while to support the most recent kernel versions.

Moschn, 2019-10-10:
Can you try to bumb the buildroot submodule and see if the 5.0.x option becomes available?

mathmax12, 2019-10-10:
@Moschn 
I didn't update my host kernel.
The kernel used in the buildroot is from [here ](https://github.com/pulp-platform/ariane-sdk/blob/1ee3c6179fd061b541bd7d8a4583699fcb34f87e/configs/buildroot_defconfig#L19), which is 5.0 x.

Maybe I changed some setting...

But it's resolved by using the latest verison of Buildroot, which has the 5.0.x.

closed 2019-10-10 by mathmax12