openhwfoundation/cva6-sdk #25: Add application to the rootfs

mathmax12, 2019-10-11:
Hi,
I tried to add an application to the rootfs, by doing which  when the Linux system boot up the corsss-compiled application can be found and executed.

I notice the [Line 92](https://github.com/pulp-platform/ariane-sdk/blob/1ee3c6179fd061b541bd7d8a4583699fcb34f87e/Makefile#L92) in Makefile gives an example for this.

So after compiled my application with **riscv64-unknown-linux-gnu-gcc** I got a binary file. Then I copied this binary file to the ./rootfs, which is used to overlay the rootfs generated in the Buildroot.

After run the ""make vmlinux" and "make bbl.bin" I got bbl.
But after the linux OS boot up I can not find the binary file that I put to the rootfs. However, the **tetris** and **cachtest** are in the system.

Did I do something wrong?

Moschn, 2019-10-11:
Your steps should work... did you run `make clean` before? maybe he uses a cached rootfs.cpio

If you want to put your own executable onto the fpga you could also just put it into the second partition of the FPGA. The second partition should get mounted on boot.

mathmax12, 2019-10-11:
I checked the buildroot/output/images/rootfs.cpio generated from buildroot.
I can see my application is there
[image]
But after the bbl used for the system booting, I didn't find that application in the system on ariane: 
[image]
.

Moschn, 2019-10-14:
Are you sure you are booting the correct bbl? If you want you could boot your vmlinux using spike or qemu and then check there if your executable is there. I suspect that somehow you are not booting the correct image.

mathmax12, 2019-10-14:
The correct bbl is used. Confirmed that!
I wonder the process of adding a new application to the rootfs is 
1. cross-compile the application go generate the binary file
2. copy the binary file to the roofs which will be used to overlay the rootfs from the Buildroot.

Do we need anything else?

mathmax12, 2019-10-15:
Thanks.
Here it is:

[bbl.zip](https://github.com/pulp-platform/ariane-sdk/files/3731639/bbl.zip)

Moschn, 2019-10-16:
You seem to have made multiple changes to the build image. I cannot boot it in qemu or spike. It hangs at `ifconfig: SIOCSIFHWADDR: No such device`.

I know that the flow in this repo did not work for you before but I cannot really help you with the lowrisc flow. I am preparing an update to the ariane-sdk repo though so maybe it will work then