openhwfoundation/cva6-sdk #27: Mount second partition

SJFeldt, 2019-11-04:
Hello,
 
I try to mount a second partition inside the provided linux system. The installation was done according to the instruction at [https://github.com/pulp-platform/ariane](Ariane Git).
 
According to this page https://github.com/pulp-platform/ariane-sdk/issues/25 the second partition on the SD card should be mounted at boot time. Unfortunately, I can't find the second partition.
 
During my search for a solution I found that Ariane repairs the GPT table of the SD card in such a way that fdisk detects the primary GPT table as corrupted, if the SD card is again inserted into my normal computer.

The system is booted from a SD card with two partitions and fdisk provides the following output:
 ```
 $ sudo sgdisk --clear
--new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdb):
$ sudo fdisk -l /dev/sdb
Disk /dev/sdb: 29,7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: B40F3443-EC5F-4185-8F91-C8E263A0DABA

Device     Start      End  Sectors  Size Type
/dev/sdb1   2048    67583    65536   32M ONIE boot
/dev/sdb2  67584 62333918 62266335 29,7G Linux filesystem
```

During the linux boot process on Ariane:
```
[   30.766351] GPT: Primary header thinks Alt. header is not at the end of the disk.
[   30.766351] GPT: Primary header thinks Alt. header is not at the end of the disk.
[   30.784484] GPT:62333951 != 62412799
[   30.784484] GPT:62333951 != 62412799
[   30.801305] GPT:Alternate GPT header not at the end of the disk.
[   30.801305] GPT:Alternate GPT header not at the end of the disk.
[   30.818495] GPT:62333951 != 62412799
[   30.818495] GPT:62333951 != 62412799
[   30.834014] GPT: Use GNU Parted to correct GPT errors.
[   30.834014] GPT: Use GNU Parted to correct GPT errors.
[   30.851527]  mmcblk0: p1 p2
[   30.851527]  mmcblk0: p1 p2
```

Afterwards I'm not able to find the second partition. Looking at the SD card at my normal computer shows me that the primary GPT table is now corrupted:
```
$ sudo fdisk -l /dev/sdb
The primary GPT table is corrupt, but the backup appears OK, so that will be used.
Disk /dev/sdb: 29,7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: B40F3443-EC5F-4185-8F91-C8E263A0DABA

Device     Start      End  Sectors  Size Type
/dev/sdb1   2048    67583    65536   32M ONIE boot
/dev/sdb2  67584 62333918 62266335 29,7G Linux filesystem
```
Using the -v of gdisk provides the following output:
```
    Problem: main GPT header's backup LBA pointer (62412799) doesn't
    match the backup GPT header's current LBA pointer (62333951).
    The 'e' option on the experts' menu may fix this problem.
    
    Problem: main GPT header's last usable LBA pointer (62412766) doesn't
    match the backup GPT header's last usable LBA pointer (62333918)
    The 'e' option on the experts' menu can probably fix this problem.
    
    Problem: Disk is too small to hold all the data!
    (Disk size is 62333952 sectors, needs to be 62412800 sectors.)
    The 'e' option on the experts' menu may fix this problem.
    
    Problem: GPT claims the disk is larger than it is! (Claimed last usable
    sector is 62412766, but backup header is at
    62412799 and disk size is 62333952 sectors.
    The 'e' option on the experts' menu will probably fix this problem
```

Running the e command fixes the error messages on my computer and brings me back to the beginning of the post.

Any idea why the second partition is not recognised? Or more precisely, why the GPT table gets corrupted?

Thanks,
Jakob

Moschn, 2019-11-04:
The bootloader should only read from the SD card and never write. It seems like this happens when booting Linux. I have never seen anything like that.

Did you use the image from the release or did you build your own? Is this repeatable? I.e., does this happen every time you boot?

SJFeldt, 2019-11-05:
Thanks for the quick reply.

I did use the image from the release (OpenPiton 12 + Ariane 4.2 release) and it is highly reliable, so it happens every time I boot.

monrick0303, 2019-11-26:
Hello,
can you tell me how about mount second partition?
did you use mount or dd?

Moschn, 2019-11-26:
I must have been mistaken. The second partition does not get automatically mounted on boot. To mount the second partition, one has to first format the second partition on the host and then mount it manually on the FPGA.

SJFeldt, 2019-11-26:
The problem seems to occur since the formatting of the SD card was done in a virtual machine.
If done on a native linux system the second partition can be mounted with a normal mount command.

Moschn, 2019-11-26:
Thanks for the update. May I close this then?

closed 2019-11-26 by SJFeldt