openhwfoundation/cva6-sdk #28: mount second partition about lowrisc rootfs

monrick0303, 2019-11-22:
Hello,

I try to mount **rootfs.tar.xz** of lowrisc (ariane-v0.7) to second partition.
following is my step:
`sudo mkdir /mnt/sdc2`
`sudo chmod 777 /mnt/sdc2`
`sudo mount -t ext4 /dev/sdc2 /mnt/sdc2`
and decompress **rootfs.tar.xz** to /mnt/sdc2

During the linux boot process on Ariane:
`[   28.498808] vmap allocation for size 6592671433917067264 failed: use vmalloc=<size> to increase size`
`[   28.498808] vmap allocation for size 6592671433917067264 failed: use vmalloc=<size> to increase size`
`[   28.527987] ioremap returned NULL`
`[   28.527987] ioremap returned NULL`
`[   28.565207] xilinx_spi 20000000.xps-spi: at 0x20000000 mapped to 0x(____ptrval____), irq=2`
`[   28.565207] xilinx_spi 20000000.xps-spi: at 0x20000000 mapped to 0x(____ptrval____), irq=2`
`[   28.619995] libphy: Fixed MDIO Bus: probed`
`[   28.619995] libphy: Fixed MDIO Bus: probed`
`[   28.648769] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000`
`[   28.648769] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000`
`[   28.709397] libphy: GPIO Bitbanged LowRISC: probed`
`[   28.709397] libphy: GPIO Bitbanged LowRISC: probed`
`[   28.720841] Probing lowrisc-0:01 (address 1)`
`[   28.720841] Probing lowrisc-0:01 (address 1)`
`[   28.741694] RTL8211E Gigabit Ethernet lowrisc-0:01: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=lowrisc-0:01, irq=POLL)`
`[   28.741694] RTL8211E Gigabit Ethernet lowrisc-0:01: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=lowrisc-0:01, irq=POLL)`
`[   28.802770] lowrisc-eth 30000000.lowrisc-eth: Lowrisc Ether100MHz registered`
`[   28.802770] lowrisc-eth 30000000.lowrisc-eth: Lowrisc Ether100MHz registered`
`[   28.825195] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver`
`[   28.825195] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver`
`[   28.900476] mmc_spi spi0.0: SD/MMC host mmc0, no WP, no poweroff, cd polling`
`[   28.900476] mmc_spi spi0.0: SD/MMC host mmc0, no WP, no poweroff, cd polling`
`[   28.930543] usbcore: registered new interface driver usbhid`
`[   28.930543] usbcore: registered new interface driver usbhid`
`[   28.942982] usbhid: USB HID core driver`
`[   28.942982] usbhid: USB HID core driver`
`[   29.013423] NET: Registered protocol family 10`
`[   29.013423] NET: Registered protocol family 10`
`[   29.095902] Segment Routing with IPv6`
`[   29.095902] Segment Routing with IPv6`
`[   29.109387] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver`
`[   29.109387] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver`
`[   29.155973] NET: Registered protocol family 17`
`[   29.155973] NET: Registered protocol family 17`
`[   29.192901] Key type dns_resolver registered`
`[   29.192901] Key type dns_resolver registered`
`[   29.770719] Freeing unused kernel memory: 6724K`
`[   29.770719] Freeing unused kernel memory: 6724K`
`[   29.780877] This architecture does not have kernel memory protection.`
`[   29.780877] This architecture does not have kernel memory protection.`
`[   29.800453] Run /init as init process`
`[   29.800453] Run /init as init process`
`[   29.877914] mmc0: host does not support reading read-only switch, assuming write-enable`
`[   29.877914] mmc0: host does not support reading read-only switch, assuming write-enable`
`[   29.897948] mmc0: new SDHC card on SPI`
`[   29.897948] mmc0: new SDHC card on SPI`
`[   29.990469] mmcblk0: mmc0:0000 SS04G 3.69 GiB`
`[   29.990469] mmcblk0: mmc0:0000 SS04G 3.69 GiB`
`[   30.782187]  mmcblk0: p1 p2`
`[   30.782187]  mmcblk0: p1 p2`
`Starting logging: OK`
`Initializing random number generator... [   37.160335] random: dd: uninitialized urandom read (512 bytes read)`
`[   37.160335] random: dd: uninitialized urandom read (512 bytes read)`
`done.`
`Starting rpcbind: OK`
`[   40.385464] random: ssh-keygen: uninitialized urandom read (32 bytes read)`
`[   40.385464] random: ssh-keygen: uninitialized urandom read (32 bytes read)`
`Starting sshd: [   40.949599] random: sshd: uninitialized urandom read (32 bytes read)`
`[   40.949599] random: sshd: uninitialized urandom read (32 bytes read)`
`OK`
`NFS preparation skipped, OK`

`Welcome to Buildroot`
`buildroot login: root`
`# ls`
`# cd ..`
`# ls`
`README.md      init           mnt            sbin           var`
`bin            lib            opt            sys`
`cachetest.elf  lib64          proc           tetris`
`dev            linuxrc        root           tmp`
`etc            media          run            usr`
`# cd bin`
`# ls`
`arch           echo           linux32        nuke           sh`
`ash            ed             linux64        pidof          sleep`
`base64         egrep          ln             ping           stat`
`busybox        false          login          pipe_progress  stty`
`cat            fatattr        ls             printenv       su`
`chattr         fdflush        lsattr         ps             sync`
`chgrp          fgrep          mkdir          pwd            touch`
`chmod          fsync          mknod          resume         true`
`chown          getopt         mktemp         rev            umount`
`cp             grep           more           rm             uname`
`cttyhack       hostname       mount          rmdir          usleep`
`date           ionice         mountpoint     run-parts      vi`
`dd             iostat         mpstat         scriptreplay   watch`
`df             kbd_mode       mt             setarch`
`dmesg          kill           mv             setpriv`
`dumpkmap       link           nice           setserial`

It looks successful, but i find't the content about rootfs of **lowrisc**
I don't know why.
Did I do something wrong?

bbl use two version on test:
1. from the release (OpenPiton 12 + Ariane 4.2 release).
2. build from ariane-sdk repository.

Thanks You.

Moschn, 2019-11-26:
In our images, the rootfs is just an initramfs. So you would need to recompile your image every time your rootfs changes. 

You could also just mount the second partition on the SD card. You would need to format it first on your host, copy your files onto it, and only then mount it on the FPGA with `mount /dev/mmcblk0p2 /mnt`.

monrick0303, 2019-11-27:
I can mount with `mount /dev/mmcblk0p2 /mnt`
so, this method can't overlay origin rootfs(initramfs) with rootfs of lowrisc?

Moschn, 2019-11-27:
Mounting at runtime does NOT overlay the rootfs. If you want to change your rootfs you will need to recompile the entire image!

Initramfs is a filesystem in RAM that is backed into the initial Linux image. Usually it gets replaced later on in boot by an actual disk. However, to keep it simple we keep the initramfs during the entire runtime.

Why do you want to overlay? If you can mount, you can directly access the data in the second partition in `/mnt`

monrick0303, 2019-11-27:
After mount successful,
I attempt run `/mnt/bin/tar` for test, but error.
I think the reason is `/mnt/lib/riscv64-linux-gnu`.
[image]

Moschn, 2019-11-27:
Why do you put an extra rootfs on the second partition? You are only supposed to put additional executables there. The main rootfs is still the one in initramfs! If you want to replace this you need to recompile the linux image and change settings in buildroot. 

It obviously does not work since it is looking for a shared library but it does not find any shared library in the first rootfs.

monrick0303, 2019-11-27:
I understand,thanks you.
So if i want to replace the rootfs, i need to place new rootfs to `/ariane-sdk/rootfs` and recompile the linux image and change setting in buildroot(ex.glibc version)?

zarubaf, 2019-11-27:
@Moschn He should be able to `chroot` to the new `rootfs` right?

Moschn, 2019-11-27:
@monrick0303 Yes that should work.

@zarubaf Maybe. I am not sure if we even include chroot in our busybox. But in theory it could work.

monrick0303, 2019-12-03:
i can use chroot.
Thanks for your help.

closed 2019-12-03 by monrick0303