openhwfoundation/cva6-sdk #30: Modifying rootfs makes Ariane unbootable

Iripi97, 2020-01-27:
Hi I am trying to add some SPEC 2006 benchmarks onto the filesystem for Ariane to run them while being deployed through an FPGA (the VC707). Whenever I add the directory containing the benchmarks into the rootfs folder (ariane-sdk/rootfs) it seems to mess the booting process, I see the following:

copying boot image ……………… done!
bbl loader
 (it gets stuck at this line, after bbl loader) 

I know an alternative is to utilize the second partition of the SD card; however I cannot seem to mount the second partition with instructions found in other issue threads. When I type:
mount /dev/mmcblk0p2 /mnt
I receive the following line back:
mount: mounting /dev/mmcblk0p2 on /mnt failed: Invalid arguments

I am adding this directory to rootfs, then running make vmlinux, and then finally make bbl.bin. Am I doing this process wrong? Is there an easy way to add folders or files to the boot image that I am missing? Any response would be greatly appreciated.

Moschn, 2020-01-28:
How big is spec? Or how big is your final `bbl.bin`? The boot partition has a maximum size. The zero stage bootloader only copies the first 30mb (I think). 

The error message on mount seems like it did not find the device `/dev/mmcblk0p2` can you try `ls /dev/` to check what devices are available?

Iripi97, 2020-01-28:
Ahh that definitely makes sense why the booting process hangs up... with the addition of SPEC it makes the bbl.bin file approximately 100 Mb.

The device mmcblk0, mmcblk0p1, and mmcblk0p2 are all listed when I check the /dev directory. 

[image]
My main confusion is because I thought this was determined as a way to access the second partion:
[https://github.com/pulp-platform/ariane-sdk/issues/28](url)
But I am also concerned that because I am using a VM (Ubuntu 16.04) this might not work, as mentioned here:
[https://github.com/pulp-platform/ariane-sdk/issues/27](url)

zarubaf, 2020-01-28:
Did you format the SD Card with a file system that Linux kernel has support for (I think `ext3` but you would need to double-check)? I also remember that the busybox mount needed the file system type to mount `mount -t ext3` otherwise it would not mount the partition.

Iripi97, 2020-01-28:
@Moschn Thank you for the insight on how large the bbl.bin file can be! 
@zarubaf Thank you for the suggestion, I had actually just played around with making the second partition ext3/ext4 and it mounts successfully now.
Do you know if it would be possible to compile a RISC-V toolchain (ideally the riscv-gnu-toolchain) with this version of Linux running through Ariane? And even some of the RISC-V tools, such as Spike and pk? 
I know, after looking in the /bin/ directory it does have limited functionality but this is something we were really hoping to achieve.

Iripi97, 2020-01-28:
Or would adding a compiled toolchain to the second partition function in the same way? I realize it would be difficult to install some of the dependent packages (autoconf, automake, autotools-dev, etc.) with Ariane lacking the ethernet connection.

zarubaf, 2020-01-29:
`spike` and `pk` should be fairly easy to cross-compile to RISC-V, I would just make sure to statically link them. Including a tool-chain with this busybox environment should technically be feasible but if there is no buildroot script that builds this toolchain for you this is a fairly intricate process. I think if you really want to have something more full-fledged you will probably need to look into something distro like. In any case, compiling on the FPGA will be very, very, slow and you are much likely better off if you cross-compile your application on your host machine.

Iripi97, 2020-01-30:
I built the RISC-V tools and copied them over to the second partition. When I try to execute spike:
./spike
I get the following error messages:
line 1: ELF: not found
line 2: syntax error: unexpected ")"
This version of Linux enabled through Ariane should be able to run these tools right?
Also could you tell me how one would statically link them? I have very little experience with this topic. Lastly, I am assuming that compiling the RISC-V tools on my Ubuntu VM and simply copying them over isn't the way to get them on to Ariane and running. Am I wrong in this thinking? Any response would be greatly appreciated.

Moschn, 2020-01-31:
I guess you would have to look into the compile time configuration of spike. 

A more general question: Why do you want to execute spike on the FPGA?

Iripi97, 2020-01-31:
@Moschn We want to use spike + pk on the FPGA because it is what will actually execute the SPEC CPU2006 binaries and give us output for each test within each of the benchmarks. 
pk operates just fine (this is expected though as it is statically linked). However spike is dynamically linked with the default installation.
Would the configuration of spike be in the various shell scripts (like build.sh or build-spike-only.sh) found in the riscv-tools directory (downloaded straight from github)? I noticed there is also a "Linker" section in the Makefile.in in the riscv-isa-sim subdirectory of riscv-tools, would this be one way to accomplish building spike (statically linked)?

zarubaf, 2020-01-31:
You need `spike` + `pk` to emulate RISC-V user space on x86 (or any other architecture except RISC-V). As you are running native RISC-V this will not gain you anything except of obtaining meaningless SPEC runtimes and even longer SPEC runtimes.

Iripi97, 2020-01-31:
@zarubaf @Moschn Haha I am so sorry. I had a complete misunderstanding on how to run the SPEC CPU2006 benchmarks. I was under the impression you needed to use a tool like Spike to run them. However I understand now that this is just a simulation of how the benchmarks would run. I now know that I need to run them natively on the Linux OS booted with Ariane. Thank you for all the intuition and helpful responses, I did not mean to cause any unnecessary confusion.

closed 2020-01-31 by Iripi97