openhwfoundation/cva6-sdk #31: Functionality of BusyBox

Iripi97, 2020-02-05:
I was wondering for purposes of running SPEC on Ariane:
Can Linux running on Ariane support a C compiler? I noticed in the configuration for BusyBox it is not included. Has anyone run Ariane with a more robust version of Linux? Is it possible to utilize the same work-flow with the SDK with a version of Linux that has increased functionality, or would this conflict with the booting process?

Moschn, 2020-02-06:
Why do you need a C compiler on the FPGA to run SPEC? You should be able to crosscompile SPEC and run it on the FPGA. No need for recompilation on the FPGA itself.

Iripi97, 2020-02-06:
I am able to cross-compile the binaries that SPEC will run just fine, however the tools (such as runspec) do not work on Ariane. I am therefor stuck having to rebuild the tools, which requires various utilities like a c compiler. Has running SPEC already been done before by Ariane's group internally? If so what steps did they take to accomplish this?