openhwfoundation/cva6-sdk #34: Linux Kernel Start Address

Kendidi, 2020-07-17:
Hi,

I am trying to load a compiled ariane-sdk based vmlinux to a FPGA (ROM built with ariane.dts) via GDB.  But the physical memory address on FPGA is 0x2_0000_0000.  I wonder where & how do I change the Kernel start/boot address??  Many thanks in advance!

Moschn, 2020-07-20:
> `riscv64-unknown-elf-objcopy -S -O binary --change-addresses -0x80000000 bbl bbl.bin`

https://github.com/openhwgroup/cva6/issues/469

Kendidi, 2020-08-01:
> 
> 
> > `riscv64-unknown-elf-objcopy -S -O binary --change-addresses -0x80000000 bbl bbl.bin`
> 
> [openhwgroup/cva6#469](https://github.com/openhwgroup/cva6/issues/469)

Thanks Moschn!

I am only using vmlinux, not bbl or bbl.bin. Can I still use "riscv64-unknown-elf-objcopy" command to change start address?  Please advise.