openhwfoundation/cva6-sdk #35: Kernel Configuration

Kendidi, 2020-08-05:
Hi,

I have some questions regarding configurating the Kernel build for vmlinux.

I was under the impression that changes made in ".../ariane-sdk/configs/linux_defconfig" and in ".../ariane-sdk/configs/buildroot_defconfig" will be copied to ".../ariane-sdk/buildroot/.config" or applied somehow during "make vmlinux".  Is that how it's supposed to work? Please correct me if I am wrong.

1.  I want to add debug symbols to the build, so I added "CONFIG_DEBUG_INFO=y" to file, "linux_defconfig". I can see that the resulted vmlinux size grew a lot.  So I assume it was successful and debug symbols have added to the ELF file.  But then when I look at ".config",  I do not see that option there.

2. I want to add DTS info to the kernel, so I ran "make menuconfig" at ".../ariane-sdk/buildroot/", and added a path to option, "Kernel  --->Build a Device Tree Blob (DTB) ". After running "make vmlinux", I found that my changes made to ".config" is gone. And vmlinux file size did not seem to have changed.
I also tried adding, "#BR2_LINUX_KERNEL_CUSTOM_DTS_PATH="/home/aie/ariane-sdk/ariane.dts" to "buildroot_defconfig" but that didn't seem to do anything as well.  Would you please advise what's the proper way to config Kernel? And how to config options like BR2_LINUX_KERNEL_CUSTOM_DTS_PATH?

Many thanks in advance!

Moschn, 2020-08-07:
General comment: The linux kernel and buildroot use a local `.config` file. However this file is huge and they also provide a way to save a 'defconfig' which is basically all to the config file from a stock one. These defconfig files are a lot shorter in size. We use the defconfig files. During the compilation, the `.config` files get overwritten. So If you want to change anything, it is easier if you do it in the defconfig file. You can also make changes to linux or buildroot with menuconfig and then after that run `make savedefconfig` to export the new defconfig. You still might have to copy it to the correct location.

The detailed questions are mostly linux or buildroot related. I would advise to ask in a forum/chat for these specific tools. I sadly do not have enough experience to answer all of your questions.

4. I believe the initial stack pointer is set up by the bootloader. Independent of your openOCD or GDB session. As such you would have to change it there.

Kendidi, 2020-08-08:
I see. Thank you Moritz!  Appreciate it.