openhwfoundation/cva6-sdk #36: Debug with OpenOCD

DeltaEnvy, 2020-08-06:
Hi.
I have some questions about debug.
1:As you said, I go to this page "https://github.com/riscv/riscv-openocd", and installed the OpenOCD , but it doesn't said how to use it to gdb , could you tell me how to use it?
2:OpenOCD need configuration file for debug, but there isn't have the OpenOCD configuration file for digilent genesys 2 ,how can i get the corresponding configuration file.

Many thanks for your reply.

Moschn, 2020-08-07:
The openocd config file is in the main Ariane (or CVA6) repository.

As soon as you launch openocd and it sucessfully connects to the FPGA it outputs a port on which you can run riscv-gdb on.

DeltaEnvy, 2020-08-07:
ok, I'll try it, after tring ,I'll tell you the result .
Thanks for your reply

DeltaEnvy, 2020-08-10:
Sorry.
After trying, it will report the following error

root@delta:/home/delta/riscv-openocd# openocd -f fpga/ariane.cfg
Open On-Chip Debugger 0.10.0+dev-01229-g9ed670771 (2020-08-06-14:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

please tell me how to solve this problem.

Moschn, 2020-08-10:
Just to check: are you sure you use the correct ariane image and the correct jtag usb port?

DeltaEnvy, 2020-08-10:
I'm sure that my ariane image is correct.
But jtag is configured by using vivado's I/O layout. I configured JA1_P, JA1_N, JA2_P, JA2_N to TDO, TDI, TCK, TMS respectively

Moschn, 2020-08-10:
Why do you configure your IO? The release image of ariane already has the correct ports specified.

DeltaEnvy, 2020-08-11:
Oh,this is my mistake, I will try again and get back to you,thanks