openhwfoundation/cva6-sdk #38: unable to resume hart 0

Kendidi, 2020-08-29:
Hi,

I am trying to load & run Linux on Ariane FPGA via OpenOCD and GDB.

It is now stuck around the end of head.S:

    .align 2
    .Lsecondary_park:
	    /* We lack SMP support or have too many harts, so park this hart */
	    wfi                                                 **<- Stuck here**
	    j .Lsecondary_park
    END(_start)

On GDB console:

    unable to resume hart 0
      dmstatus =0x00030c82
      was stepping, halting

There is only one hart.  I wonder what's the reason it could stuck there?  Please advise.

Thanks a lot in advance!

makermuyi, 2021-01-05:
I have the same problem as you.
`
unable to resume hart 0
  dmcontrol=0x40000001
  dmstatus =0x00030c82
  was stepping, halting
unable to halt hart 0
  dmcontrol=0xc0000001
  dmstatus =0x00000c82
Hart was not halted after single step!
unable to step rtos hart
`

CaiMaoyuan, 2023-11-02:
I met the same problem.