openhwfoundation/cva6-sdk #41: Linux Kernel can not set breakpoint with openocd and gdb on GENESYS2 board

hilihui, 2020-11-04:
Hi,

I tried to  debug Linux Kernel with openocd and gdb on GENESYS2 board(the board is FPGA).

When BBL start Linux Kernel, i set a breakpoint on Kernel entry(start_kernel).The address of the breakpoint is physical address(0x80200000),and the breakpoint can work.
But when i continue to set other breakpoints, gdb always return the following error:
[image]
Thanks a lot in advance!

Moschn, 2020-11-04:
It seems like one of your breakpoints is actually in virtual memory (0xffffffe0000006be). GDB and OpenOCD do not yet fully support virtual memory debugging so I guess you have to wait until this is fully supported.

There is some preliminary support in the newest openOCD that you can enable with some configuration in `ariane.cfg`.

hilihui, 2020-11-04:
Is the newest openocd for riscv?  How to config?Thanks!

Moschn, 2020-11-04:
I never tried it myself but take a look at https://github.com/riscv/riscv-openocd/pull/386

Moschn, 2020-11-04:
Or take a look at this: https://github.com/riscv/riscv-openocd/pull/425

hilihui, 2020-11-05:
Hi, my openocd is newest version and support virtual address. Is there other reason? I see that someone can debug linux kernel with openocd on arm cpu，but my cpu is riscv.

Moschn, 2020-11-05:
The openOCD port to risc-v did not support virtual memory a year ago. Maybe this has changed now. I do not know. Take a look at the linked PRs to the risc-v openOCD.

closed 2020-11-05 by Moschn