openhwfoundation/cva6-sdk #42: A kernel message log is output twice through the serial port

MrHaiCoder, 2020-11-13:
A message log is output twice through the serial port

[image]

zarubaf, 2020-11-13:
I guess this is because the UART driver is being loaded relatively late in the boot process, therefore duplicating any earlier messages. `dmesq` should just show you a single entry.

closed 2020-11-13 by zarubaf

MrHaiCoder, 2020-11-16:
> 我猜这是因为UART驱动程序在启动过程中相对较晚地加载，因此会复制任何较早的消息。`dmesq`应该只显示一个条目。

dmesg  does display only one entry  and the application outputs an entry;
but The same is true when loading Ko driver files manually;
What could be the reason for this？

I am use gensys2 fpga board.

[image]

MrHaiCoder, 2020-12-30:
> 我猜这是因为UART驱动程序在启动过程中相对较晚地加载，因此会复制任何较早的消息。`dmesq`应该只显示一个条目。

The printk in the later driver will be like this. Can you fix it?