openhwfoundation/cva6-sdk #46: I can not log in to linux by default username root when starting the operating system

Wcm926, 2021-04-07:
Nothing happens when I type root , and the log is as follows

[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 5.1.0-rc7 (michscha@toscana.ee.ethz.ch) (gcc version 8.2.0 (GCC)) #2 SMP Tue Jun 4 23:34:11 CEST 2019
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] software IO TLB: mapped [mem 0xbb1fd000-0xbf1fd000] (64MB)
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 15 pages/cpu s24280 r8192 d28968 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 952132K/1046528K available (3710K kernel code, 225K rwdata, 1046K rodata, 6727K init, 797K bss, 94396K reserved, 0K cm)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] plic: mapped 3 interrupts with 1 handlers for 2 contexts.
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x6eb3e4560, max_idle_ns: 881590406254 ns
[    0.000353] sched_clock: 64 bits at 15MHz, resolution 66ns, wraps every 4398046511080ns
[    0.322857] printk: console [hvc0] enabled
[    0.330378] Calibrating delay loop (skipped), value calculated using timer frequency.. 30.00 BogoMIPS (lpj=150000)
[    0.347895] pid_max: default: 32768 minimum: 301
[    0.365323] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.376823] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.415649] *** VALIDATE proc ***
[    0.470625] rcu: Hierarchical SRCU implementation.
[    0.504449] smp: Bringing up secondary CPUs ...
[    0.511317] smp: Brought up 1 node, 1 CPU
[    0.538613] devtmpfs: initialized
[    0.599438] random: get_random_bytes called from setup_net+0x32/0x15c with crng_init=0
[    0.618473] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.634829] futex hash table entries: 256 (order: 2, 16384 bytes)
[    0.665528] NET: Registered protocol family 16
[    1.106585] usbcore: registered new interface driver usbfs
[    1.117690] usbcore: registered new interface driver hub
[    1.128637] usbcore: registered new device driver usb
[    1.177472] clocksource: Switched to clocksource riscv_clocksource
[    1.253910] NET: Registered protocol family 2
[    1.293509] tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 8192 bytes)
[    1.307995] TCP established hash table entries: 8192 (order: 4, 65536 bytes)
[    1.323435] TCP bind hash table entries: 8192 (order: 5, 131072 bytes)
[    1.341573] TCP: Hash tables configured (established 8192 bind 8192)
[    1.362747] UDP hash table entries: 512 (order: 2, 16384 bytes)
[    1.373308] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes)
[    1.391799] NET: Registered protocol family 1
[    1.431964] RPC: Registered named UNIX socket transport module.
[    1.441253] RPC: Registered udp transport module.
[    1.448968] RPC: Registered tcp transport module.
[    1.456996] RPC: Registered tcp NFSv4.1 backchannel transport module.
[   22.714990] workingset: timestamp_bits=62 max_order=18 bucket_order=0
[   23.332365] NFS: Registering the id_resolver key type
[   23.341252] Key type id_resolver registered
[   23.347356] Key type id_legacy registered
[   23.353888] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
[   23.831282] io scheduler mq-deadline registered
[   23.839212] io scheduler kyber registered
[   26.951121] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[... 184 lines cut ...]
[   29.791386] This architecture does not have kernel memory protection.
[   29.810886] Run /init as init process
[   29.810886] Run /init as init process
[   29.885485] mmc0: host does not support reading read-only switch, assuming write-enable
[   29.885485] mmc0: host does not support reading read-only switch, assuming write-enable
[   29.905690] mmc0: new SDXC card on SPI
[   29.905690] mmc0: new SDXC card on SPI
[   29.998184] mmcblk0: mmc0:0000 SD128 119 GiB 
[   29.998184] mmcblk0: mmc0:0000 SD128 119 GiB 
[   30.788633]  mmcblk0: p1 p2
[   30.788633]  mmcblk0: p1 p2
Starting logging: OK
Initializing random number generator... [   37.165131] random: dd: uninitialized urandom read (512 bytes read)
[   37.165131] random: dd: uninitialized urandom read (512 bytes read)
done.
Starting rpcbind: OK
[   40.402161] random: ssh-keygen: uninitialized urandom read (32 bytes read)
[   40.402161] random: ssh-keygen: uninitialized urandom read (32 bytes read)
Starting sshd: [   40.962577] random: sshd: uninitialized urandom read (32 bytes read)
[   40.962577] random: sshd: uninitialized urandom read (32 bytes read)
OK
NFS preparation skipped, OK

Welcome to Buildroot
buildroot login:

zarubaf, 2021-04-07:
Do you see the characters you type?

Wcm926, 2021-04-08:
I can not see them. I hit the enter key and nothing happens

Wcm926, 2021-04-08:
It is solved by replacing the serial communication tool.

closed 2021-04-08 by Wcm926