openhwfoundation/cva6-sdk #47: After booting of kernel, can I locate dhrystone stored in the SD card in arianne?

Wcm926, 2021-04-11:
Could you tell me how to mount the second partition of the SD Card? Thanks!

Moschn, 2021-04-12:
I think you should be able to just mount it using normal linux commands (i.e., `mount`)

leemango1998, 2021-04-12:
`mount /dev/mmcblk0p2 /mnt`
then you can see the second partition of the SD Card in `/mnt`
By the way, it's said that performance_counter module  has not fully implement the use mode on released ariane 4.2 (or before) , you may meet illegal instructions  error when running dhrystone or other benchmarks.

Wcm926, 2021-04-13:
> `mount /dev/mmcblk0p2 /mnt`
> then you can see the second partition of the SD Card in `/mnt`
> By the way, it's said that performance_counter module has not fully implement the use mode on released [...]

I tried this command yesterday, but there is an error 
# mount /dev/mmcblk0p2 /mnt
mount: mounting /dev/mmcblk0p2 on /mnt failed: Invalid argument

leemango1998, 2021-04-13:
To properly part the SD card on the host PC:
`sudo fdisk -l` # search for the corresponding disk label (e.g. /dev/sdc)
`sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdc -g` # create a new //gpt partition table and two partitions: 1st partition: 32mb (ONIE boot), second partition: rest (Linux root)
`sudo mkfs -t ext3 /dev/sdc1 ` #make filesystem ext3
`sudo mkfs -t ext3 /dev/sdc2`
`sudo dd if=bbl.bin of=/dev/sdc1 status=progress oflag=sync bs=1M `#put Linux image bbl file on sdc1
`sudo mount /dev/sdc2 /mnt` #mount sdc2 to` /mnt` and copy your files to sdc2 , do not us `dd`
`sudo cp YourFiles /mnt`
then you can mount the second part of the sd card on ariane-sdk linux.

Wcm926, 2021-04-13:
> To properly part the SD card on the host PC:
> `sudo fdisk -l` # search for the corresponding disk label (e.g. /dev/sdc)
> `sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdc -g` # create a new //gpt partition table and two partitions: 1st partition: 32mb (ONIE boot), second partition: rest (Linux root)
> `sudo mkfs -t ext3 /dev/sdc1 ` #make filesystem ext3
> `sudo mkfs -t ext3 /dev/sdc2`
> `sudo dd if=bbl.bin of=/dev/sdc1 status=progress oflag=sync bs=1M `#put Linux image bbl file on sdc1
> `sudo mount /dev/sdc2 /mnt` #mount sdc2 to` /mnt` and copy your files to sdc2 , do not use `dd`
> `sudo cp YourFiles /mnt`
> then you can mount the second part of the sd card on ariane-sdk linux.

Thank you very much! The problem is solved!

closed 2021-04-13 by Wcm926