openhwfoundation/cva6-sdk #49: The boot program in the SD card cannot start the system

2536786901, 2021-04-16:
Dear Author:
When I was performing the "Booting from an SD card" operation, I put the downloaded bbl.bin file into the SD card, and the output content when using FPGA for simulation is as follows:
[image]
It seems that the operating system has not been started. I am using FPGA for the first time. Does the bbl.bin file contain an image file? If not, can you give me a website?

KUEE1080, 2021-08-04:
> Dear Author:
> When I was performing the "Booting from an SD card" operation, I put the downloaded bbl.bin file into the SD card, and the output content when using FPGA for simulation is as follows: [...]

Did you solve this problem?? I am also experiencing the same issue.

KUEE1080, 2021-08-04:
Actually, I just solved this issue. I don't know why this works but I used balenaEtcher program to flash the SD card.

https://www.balena.io/etcher/

0ena, 2022-06-02:
I am facing this same issue as @KUEE1080 with the booting being stuck after the copying of the boot image is finished.
I am using the latest version of the CVA6-SDK directory.
Any thoughts on how to resolve this issue??

Thank you in advance for your responses!

eyssartk, 2022-06-03:
How many partition do you see on your SD Card ?
Can you dump the beginning of the first partition and compare it to fw_payload.bin ? They should be identical.

0ena, 2022-06-03:
The **last lba** and **first lba** discrepancies have to do with the one SD being 16GB and the other 32GB.
So that is irrelevant.

To be honest I have no idea what the issue causing this was.
I installed the new CVA6_SDK directory and the images created were not able to pass to the bootloader stage.
I also created images with the older Ariane-SDK directory, that used the bbl and still the images failed to pass bootloader stage.

Having tried everything, I decided to reboot my server.
This solved the issue as I am now able to create images that can boot perfectly fine, either with the old bootloader (bbl) or the new one (sbi).

Anyway, thank you @eyssartk for your help.

eyssartk, 2022-06-06:
The first partition starts at 800, the second one starts at 100000, whatever the size of the SD card.

closed 2022-10-07 by eyssartk