openhwfoundation/cva6-sdk #65: Booting 32-bit Linux on 32-bit CVA6

Iripi97, 2022-04-07:
Successfully built the 32-bit bitstream for the Xilinx VC707 FPGA implementation of CVA6. Upon building the SDK values and adding them to the SD card I get the following:
[image]
Did I mess something up? Is this something anyone has encountered before? Any response would be greatly appreciated.

eyssartk, 2022-04-08:
Hello,

We did not test Linux boot on VC707.
It looks like the kernel image on your SD card is not correct or not read correctly.
To confirm that, can you try to copy the uImage through JTAG ?

Regards
Kevin Eyssartier

Iripi97, 2022-04-08:
@eyssartk 
Could you clarify what you mean by "can you try to copy the uImage through JTAG?"
I followed all the steps the repo says (used the Makefile to format and flash the sdcard).

eyssartk, 2022-04-08:
From your log, I can see that u-boot loads correctly. The problem arise when u-boot wants to copy and load the uImage (the linux image) from the SD-card. The problem can be during the uImage generation or during the write or read of the SD-card. If you successfully boot the install32/uImage file to the board by another mean (ie JTAG) we can rule out the generation problem.

Iripi97, 2022-04-08:
@eyssartk 
I am not sure how to do this, even after looking up examples on the internet... how would this be done for this project?

eyssartk, 2022-04-08:
You will need to install openocd and riscv-gdb. For that you should add in configs/buildroot32_defconfig
```
BR2_PACKAGE_HOST_OPENOCD=y
BR2_PACKAGE_HOST_GDB=y
```
then rebuild with 
```
make images
```
You should now have buildroot/output/host/bin/openocd and buildroot/output/host/riscv32-linux-gdb.
Plug the JTAG connector and power on the board.
You can launch openocd with 
```
$(CVA6-SDK-PATH)/buildroot/output/host/bin/openocd -f $(CVA6-PATH)/corev_apu/fpga/ariane.cfg
```
then you can connect to it with 
```
$(CVA6-SDK-PATH)/buildroot/output/host/bin/riscv32-linux-gdb -ex "target remote :3333" -ex "restore $(CVA6-SDK-PATH)/install32/uImage binary 0xA0000000 -ex "continue"
```
Finally, in the u-boot serial terminal you can boot with 
```
setenv fdt_high 0xFFFFFFFF
bootm 0xA0000000 - $(fdtcontroladdr)
```

sjthales, 2022-04-08:
Hello, I think you Linux image is bigger than the amount of bytes transferred from SD card to DDR:
 - There are 28672 blocks of 512 bytes transferred = 14680064 bytes
 - your image has a size of 14904689 bytes 
 - So your Linux image is not entirely read, that explains the CRC error. 

Could you try to modify the file : **cva6-sdk/u-boot/configs/openhwgroup_cv32a6_genesysII_defconfig**

- By replacing the line:
CONFIG_BOOTCOMMAND="mmc info; mmc read 90000000 100000 7000; setenv fdt_high 0xffffffff; bootm 90000000 - $(fdtcontroladdr)"
- by 
CONFIG_BOOTCOMMAND="mmc info; mmc read 90000000 100000 **8000**; setenv fdt_high 0xffffffff; bootm 90000000 - $(fdtcontroladdr)"

Iripi97, 2022-04-08:
@sjthales Thank you very much this solved the issue of getting the image to boot. Now however the image gets stuck booting Linux:
[image]
The last time this happened (with older 64-bit Ariane/CVA6 using the bbl.bin file for booting) the solution was to modify the device tree file for generating the bitstream (what would be /cva6/corev_apu/bootrom/ariane.dts for the updated version). I notice there are "eth" definitions all over this file, is that causing the issue?
Or would bypassing the ethernet hang up be accomplished by modifying _openhwgroup_cv32a6_genesysII_defconfig_ configuration file (cva6-sdk/u-boot/configs) since it has a line:
CONFIG_DEFAULT_DEVICE_TREE="cv32a6_genesysII"

eyssartk, 2022-04-11:
As I told you, we did not test the Linux boot in the VC707. The used device tree is located in u-boot/arch/riscv/dts/cv##a6_genesys.dts. You can modify it or create another one for the VC707.

Iripi97, 2022-04-15:
To anyone else attempting to Test CV32A6 on the VC707, the following changes are required:
(1) Modify the line in cva6-sdk/u-boot/configs/openhwgroup_cv32a6_genesysII_defconfig to say
_CONGIF_BOOTCOMMAND="mmc info; mmc read 90000000 100000 8000; setenv fdt_high 0xffffffff; bootm 90000000 -$(fdtcontroladdr)"_
(2) Remove the ethernet definition in cva6-sdk/u-boot/arch/riscv/dts/cv32a6_genesysII.dts
[image]

@eyssartk @sjthales
Is the procedure for adding custom applications to the rootfs the same? (Copy the compiled binary/executable to cva6-sdk/rootfs and regenerate the images)
Are there any restrictions to how much I can add to the rootfs? I know previously (when SDK used bbl.bin only) the rootfs could not be larger than approx. 20 MB.

eyssartk, 2022-04-26:
Yes, you can add pre-compiled application in the rootfs folder. 
The BR2_TARGET_ROOTFS_EXT2_SIZE specify the maximum size of your initramfs.
This can affect the uImage size, it is possible that you will need to change CONFIG_BOOTCOMMAND again.

closed 2022-05-19 by Iripi97