openhwfoundation/cva6-sdk #80: make flash-sdcard can not create partitions. (Tried with multiple microSD cards)

arash-vafaei, 2023-06-30:
HI there team, 
I am trying to build Ariane on a Gensys2 board. I have built the images successfully and spike simulation works with the image. I just have problems when trying to flash the images into the two partitions on an sd card. basically, the partitions do not get created with "sudo -E make flash-sdcard SDDEVICE=/dev/sdb"

the final command is:
sgdisk --clear -g --new=1:2048:6909 --new:2:512M:0 --typecode=1:3000 --typecode=2:8300 /dev/sdb

the error is :
could not create partition 1 from 2048 to 6909
could not create partition 2 from 1048576 to 2014
could not change partition 1's type code to 3000
could not change partition 2's type code to 8300
Error encountered not saving changes.
[makefile:142 format-sd ] Error4

I tried with new microSDs from different vendors and still the same problem arises.
Any Ideas?

eyssartk, 2023-07-03:
Hello,

Can you confirm with the command "lsblk /dev/sdb" that a device is present ?

Regards,

arash-vafaei, 2023-07-03:
So it returns:
"not a block Device"

Also When I try "gdisk" and "w" command to write a GPT table it gives me:

"OK; writing new GUID partition table (GPT) to /dev/sdb.
Unable to open device '/dev/sdb' for writing! Errno is 13! Aborting write!"

eyssartk, 2023-09-18:
It seems your host does not recognize the device. 
Are you still stuck on this problem ?