openhwfoundation/cva6-sdk #82: My CVA6-SDK won't boot up. Why?

DonierZhang, 2023-08-23:
I follow the steps came from segment: Booting SMP Linux on Genesys2 or VC707 in link https://github.com/PrincetonUniversity/openpiton
downloaded bbl.bin from: https://github.com/openhwgroup/cva6-sdk/releases/tag/v0.3.0-op 

and burned it to the SD card as follows:
sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdb
sudo dd if=bbl.bin of=/dev/sdb1 oflag=sync bs=1M

Then I inserted the SD card into the FPGA development board and it ran fine.

After that I wanted to compile a version under CVA6-SDK for testing.

I did all the steps under ubuntu as described in Quickstart:

step 1. sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev libusb-1.0-0-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev device-tree-compiler pkg-config libexpat-dev

step 2. git submodule update --init --recursive

step 3. make images

At the end of the compilation, I can see the corresponding files generated in the install64/:
 fw_payload.bin fw_payload.elf Image Image.gz u-boot.bin uImage vmlinux

Then I started to execute:
step 4. sudo -E make flash-sdcard SDDEVICE=/dev/sdb

messages showing this execution was successful.

After that, I inserted the SD card into the FPGA development board and restart board but after loading block, system is blocked with the following message:
copying boot image
copying block 32700 of 32768 blocks(%99)
  done !

Is there any step missing or wrong in my above operation under cva6-sdk?

eyssartk, 2023-09-18:
Hello,

I didn't test the cva6-sdk with the Openpiton configuration. Can you open this issue on their repository ? You will probably have better answers.

Regards

dassheladiya, 2025-03-17:
I am experiencing the same issue. Have there been any updates on this?

@DonierZhang did you open a issue on OpenPiton repository?