openhwfoundation/cva6-sdk #84: Error while booting using OpenOCD

jimaandro, 2023-09-18:
I have load a linux image created with cva6-sdk to the Ariane. I used OpenOCD to load the image, then I tried to verify_image but I got
`> verify_image uImage 0x80200000
Algorithm timed out after 200001 ms.
ra = 0x0
sp = 0x0
gp = 0x0
tp = 0x0
t0 = 0x0
t1 = 0x0
t2 = 0x0
s0 = 0x0
s1 = 0x0
a0 = 0x80200000
a1 = 0x98b3bf
a2 = 0x0
a3 = 0x0
a4 = 0x0
a5 = 0x0
a6 = 0x0
a7 = 0x0
s2 = 0x0
s3 = 0x0
s4 = 0x0
s5 = 0x0
s6 = 0x0
s7 = 0x0
s8 = 0x0
s9 = 0x0
s10 = 0x0
s11 = 0x0
t3 = 0x0
t4 = 0x0
t5 = 0x0
t6 = 0x0
pc = 0x80200040
mstatus = 0xa00001800
mepc = 0x80200040
mcause = 0xc
error executing RISC-V CRC algorithm`
Then I checked this address and there was an invalid instruction 0x08088b1f. So why does sdk produce this?

eyssartk, 2023-09-18:
Can you dump the uImage with 
```objdump -S uImage```
And check what is the instruction at 0x80200040 ?

mcause = 0xc means Instruction page fault.

Regards

jimaandro, 2023-09-18:
Thank you very much for the reply.
Here is a part of the dump image in attached files.
[uImage.zip](https://github.com/openhwgroup/cva6-sdk/files/12648412/uImage.zip)

Also, sometimes it shows mcause = 0x2 (Illegal instruction)

closed 2023-09-23 by jimaandro