openhwfoundation/cva6-sdk #85: cva6-sdk uart driver doesn't work for CVA6

jimaandro, 2023-09-23:
### Is there an existing CVA6-SDK task for this?

- [X] I have searched the existing task issues

### Task Description

I am trying to boot the fw_payload via OpenOCD and GDB. It is executing correctly till it fails to initialize IRQ chip, (but this is not the actual problem). The problem is that before trying to initialize the IRQ it should have printed the OpenSBI banner and it doesn't

	sbi_boot_print_banner(scratch);

	rc = sbi_platform_irqchip_init(plat, TRUE);
	if (rc) {
		sbi_printf("%s: platform irqchip init failed (error %d)\n",
			   __func__, rc);
		sbi_hart_hang();
	}

I also noticed that the [driver of OpenSBI](https://github.com/riscv-software-src/opensbi/blob/be245acfffa297b5ed4e0c7bb473a6bd55231bf8/lib/utils/serial/uart8250.c) is a bit different from CVA6 [uart drivers](https://github.com/openhwgroup/cva6/blob/master/corev_apu/fpga/src/bootrom/src/uart.c) and also it doesn't match with the [uart hardware](https://github.com/pulp-platform/apb_uart/blob/ac3461ce23832e02903b9d2d7d4edd7fcb89bd92/src/apb_uart.vhd) (UART 16750)
So what should I do to print in uart?

### Description of Done

I tried to change the UART IP iin my disign to UART 16550 in order to match with the driver, but it didn't work

eyssartk, 2023-09-25:
Hello,

With GDB + OpenOCD, you remove the arguments passed by the bootrom to OpenSBI firmware. 
Register a1 needs to contain the address of the device tree binary. There is one DTB available in the bootrom but its adress is dependent of the bootrom compilation. 

You can check the address of the bootrom's DTB in https://github.com/openhwgroup/cva6/tree/master/corev_apu/fpga/src/bootrom
by compiling the bootrom and checking the _dtb symbol's address. The DTB should start with 0xd00dfeed. On my machine, I find address 0x10C00 for 64bits bootrom and 0x11200 for 32bits.

Or you can load your own dtb somewhere in ram.

Good luck !
Kevin

jimaandro, 2023-10-01:
Thank you very much for the reply. Well, the problem actually was on PLIC, I had to reduce input sources from 30 to 1 ( I use only UART, so I have only this interrupt). Now it prints the OpenSBI logo and booting, and then it goes to u-boot. The problem is that it doesn't print everything correctly (e.g. it misses characters) like this:
[image]

closed 2024-04-17 by jimaandro