openhwfoundation/cva6-sdk #99: Trouble Accessing CVA6 via SSH

mrbilandi, 2024-07-08:
Hello,

I have synthesized the CVA6 RTL code for Genesys2 as described in the README file, built the Linux image using the CVA6-SDK, and now I have Linux running on the CVA6 on our board. I set a static IP for eth0, and I can ping the board from a remote machine. However, I can't SSH to it. Whenever I try to SSH to the board, it asks for a password. Since there is no password, I just press enter, but the message 'Permission denied, please try again.' appears. I also tried setting a password like 'root', but the same thing happens: after entering the password(root), it responds with 'Permission denied, please try again.'

Is it supposed to be possible to SSH to the board from a remote machine?

mrbilandi, 2024-07-08:
Just to mention, whenever Linux boots, I don't see the welcoming message "Welcome to Buildroot," and it eventually logs in without any user input. The Linux boot-up process is as follows:

Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD...
SD command cmd0 response : 01
SD command cmd55 response : 01
SD command cmd41 response : 01
SD command cmd55 response : 01
SD command cmd41 response : 00
sd initialized!
gpt partition table header:
signature: 5452415020494645
revision: 00010000
size: 0000005C
crc_header: 84D98092
reserved: 00000000
current lba: 0000000000000001
backup lda: 0000000001CEA7FF
partition entries lba: 0000000000000002
number partition entries: 00000080
size partition entries: 00000080
gpt partition entry 00
partition type guid: D5F7127456A1134B81DC867174929325
partition guid: 895830B2203A304580EB1BCE0ADF5A73
first lba: 0000000000000800
last lba: 0000000000001AFD
attributes: 0000000000000000
name: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 01
partition type guid: AF3DC60F838472478E793D69D8477DE4
partition guid: D3C040C5431AA045A91CC4EB5F1C16A3
first lba: 0000000000100000
last lba: 0000000001CEA7DE
attributes: 0000000000000000
name: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 02
partition type guid: 00000000000000000000000000000000
partition guid: 00000000000000000000000000000000
first lba: 0000000000000000
last lba: 0000000000000000
attributes: 0000000000000000
name: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 03
partition type guid: 00000000000000000000000000000000
partition guid: 00000000000000000000000000000000
first lba: 0000000000000000
last lba: 0000000000000000
attributes: 0000000000000000
name: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
copying boot image .... done!

OpenSBI v0.9

/ __ \ / | _ _ |
| | | | __ ___ _ __ | ( | |) || |
| | | | '_ \ / _ \ '_ \ ___ | _ < | |
| || | |) | __/ | | |) | |) || |
_/| ./ _|| ||/|____/|
| |
|_|

Platform Name : ARIANE RISC-V
Platform Features : medeleg
Platform HART Count : 1
Platform IPI Device : aclint-mswi
Platform Timer Device : aclint-mtimer @ 1000000Hz
[... 270 lines cut ...]
[ 19.547475] mmcblk0: mmc0:0000 SD16G 14.5 GiB
[ 20.177821] mmcblk0: p1 p2
[ 20.177821] mmcblk0: p1 p2
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [ 25.629738] random: dd: uninitialized urandom read (512 bytes read)
[ 25.629738] random: dd: uninitialized urandom read (512 bytes read)
OK
Starting rpcbind: OK
[ 27.084800] Open device, request interrupt 11
[ 27.084800] Open device, request interrupt 11
[ 31.694763] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 31.694763] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 33.415042] random: crng init done
[ 33.415042] random: crng init done
Starting sshd: OK
NFS preparation skipped, OK

ls
README.md init mnt sbin var
bin lib opt sys
cachetest.elf lib64 proc tetris
dev linuxrc root tmp
etc media run usr

Jorgebrihu, 2025-03-04:
Were you able to communicate with the cva6 through ssh? 
I have been also asked for the password.
Is there a way to modify the linux configuration, so I am not asked for a password and then rebuild the linux image?

mrbilandi, 2025-03-05:
Hi @Jorgebrihu 
Yes, I was able to communicate with the CVA6 through SSH. I will upload the code and the procedure very soon and will let you know.