openhwfoundation/cva6-sdk #101: [TASK] <title> Running "make images" command using the docker container doesn't work.

walrider3, 2024-07-27:
### Is there an existing CVA6-SDK task for this?

- [X] I have searched the existing task issues

### Task Description

>>> uclibc 1.0.49 Building
/usr/bin/make  -C /repo/buildroot/output/build/uclibc-1.0.49 ARCH="riscv64" CROSS_COMPILE="/repo/buildroot/output/host/bin/riscv64-buildroot-linux-uclibc-" UCLIBC_EXTRA_CFLAGS=" -g0 -fno-lto" HOSTCC="/usr/bin/gcc" headers
make[2]: Entering directory '/repo/buildroot/output/build/uclibc-1.0.49'
make[2]: 'headers' is up to date.
make[2]: Leaving directory '/repo/buildroot/output/build/uclibc-1.0.49'
/usr/bin/make  -C /repo/buildroot/output/build/uclibc-1.0.49 ARCH="riscv64" CROSS_COMPILE="/repo/buildroot/output/host/bin/riscv64-buildroot-linux-uclibc-" UCLIBC_EXTRA_CFLAGS=" -g0 -fno-lto" HOSTCC="/usr/bin/gcc"
make[2]: Entering directory '/repo/buildroot/output/build/uclibc-1.0.49'
make[3]: 'lib/ld-uClibc.so' is up to date.
  LD libuClibc-1.0.49.so
/repo/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(err.os): in function `.L0 ':
err.c:(.text+0xa0): undefined reference to `__atomic_compare_exchange_4'
/repo/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(err.os): in function `.L6':
err.c:(.text+0x138): undefined reference to `__atomic_exchange_4'
/repo/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(__cxa_finalize.os): in function `.L3':
__cxa_finalize.c:(.text+0xac): undefined reference to `__atomic_compare_exchange_8'
/repo/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(fork.oS): in function `.L23':
fork.c:(.text+0x130): undefined reference to `__atomic_fetch_add_4'
/repo/buildroot/output/host/lib/gcc/riscv64-buildroot-linux-uclibc/13.3.0/../../../../riscv64-buildroot-linux-uclibc/bin/ld: libc/libc_so.a(sem_timedwait.oS): in function `.L5':
sem_timedwait.c:(.text+0xd8): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
libc/Makefile.in:77: recipe for target 'lib/libc.so' failed
make[2]: *** [lib/libc.so] Error 1
make[2]: Leaving directory '/repo/buildroot/output/build/uclibc-1.0.49'
package/pkg-generic.mk:286: recipe for target '/repo/buildroot/output/build/uclibc-1.0.49/.stamp_built' failed
make[1]: *** [/repo/buildroot/output/build/uclibc-1.0.49/.stamp_built] Error 2
make[1]: Leaving directory '/repo/buildroot'
Makefile:76: recipe for target '/repo/buildroot/output/host/bin/riscv64-buildroot-linux-gnu-gcc' failed
make: *** [/repo/buildroot/output/host/bin/riscv64-buildroot-linux-gnu-gcc] Error 2
I have no name!@0fd45013006f:/repo$ 

### Description of Done

I need to get the files for to flash the sd card with so I can boot linux on the cva6 core on the genesys 2 board. But the command gives the following errror.

wal222, 2024-07-27:
I now get the following error, the **wget --passive-ftp -nd -t 3 -O '/repo/buildroot/output/build/.linux-v5.10.7-br1.tar.gz.hEmhDo/output' 'http://sources.buildroot.net/linux-v5.10.7-br1.tar.gz'** command failed and the file **linux-v5.10.7-br1.tar.gz** actually doesn't exist on the address it is said to exist:
[image]

Moschn, 2024-08-06:
I cannot reproduce your issues, therefore, I also cannot offer much help. The error messages that you posted point towards a broken toolchain, but the toolchain should be built by `make images` so I have no idea what is going on. 

I just did a clean build locally on ubuntu 24.04 and it worked without issues. If it does not work for you on your setup, try on a clean ubuntu image (either in docker, or a full VM).

walrider3, 2024-08-08:
> I cannot reproduce your issues, therefore, I also cannot offer much help. The error messages that you posted point towards a broken toolchain, but the toolchain should be built by `make images` so I [...]

I'm already running the command in a docker container(as I have mentioned in the title) that too on a new and clean ubuntu image I setup as dual boot.

Moschn, 2024-08-09:
Are you following all the steps outlined in the readme? Also, please do not tamper with the git submodules and their version. The current master of this repo should be working. A simple `git submodule update --init --recursive` is all you need.

jason23g, 2024-11-13:
Hello guys,

I have also encountered the same problem when  i use make clean-all and then make all in my host machine instead of docker. Is there any update available related to this error?

Thanks in advance for your time.